UofSSpaceTeam/roveberrypy #8: Clean Up and Create URC2016 Branch+Tag

ausshir, 2016-07-08:
We will need to clean up everything and prepare for a branch. The following is on this list:

- [ ] Check what is working or broken, what files are duplicated or garbage. *Please discuss below!*
- [ ] Create a branch and tag/release for URC2016
- [ ] Strip down the core rover software to the bare minimum so we can port it to python 3

Some bonus ideas
- [ ] Move the embedded (microcontroller) code into its own repository
- [ ] Crate a repository for our OS level configuration (device trees, kernel build scripts etc.)

ottopasuuna, 2016-07-08:
Coms for the arm is sort of broken, and I never got the gamepad buttons to work.
Also, do we still need hardwareImages/toFlash_edison.7z ?
I'll add Thumbs.db to the gitignore.

ausshir, 2016-07-08:
1. If we document that somewhere then that's probably fine?
2. Probably not. It would make sense to put stuff like that in the future OS repo.
3. Good idea!

ottopasuuna, 2016-07-10:
I'll make a list of the rover processes. Check off (if you can) the ones that you know are working or will only take very small changes to get working:
- [ ] ~~Arm process~~ (game pad button support not quite working)
- [ ] Camera process
- [ ] Drive process
- [ ] Drill process
- [ ] Navigation process
- [ ] Storage bin process
- [ ] Web server process
- [ ] Web server
- [ ] Can server
- [ ] Json server

Here are the embedded modules:
- [ ] ~~Arm~~ (Comunication is buggy)
- [ ] Drill
- [ ] Storage bin
- [ ] Science sensors

ausshir, 2016-07-11:
You can totally remove WebServerProcess as it should have been deprecated and repalced with WebServer a while ago.

Drive process is missing the tweaks we made on the rover before swapping out the compute module board but it was working well enough to drive and the other functionality is there.

Drill and Storage Bin process have never been tested in the field. I think the drill Teensy is fried. The python code worked in the lab and so did the Teensy code.

Camera process is totally not implemented but it was basically just going to be a bridge between the Python and the Teensy so it probably can just survive how it is with a note (with the commented code).

No idea about the Navigation Process. It gets GPS coordinates and seems to have everything it needs.

The science sensors definitely do not work right. So a note will need to be made there too.

*For all the places with a note I think we should make a README.md file for future reference.*

ausshir, 2016-07-12:
Updated some of those README files. I think it is sufficient to see what was going on - we really didn't have much code.

Unless there's any major objections we can probably create a URC 2016 Release. (In the release notes make sure to highlight the WebServer integration as well as the improvments to the way processes register with main.py - everything else seemed to be a regression! :O)

I changed the name of this issue because I think it would be more fitting to create a new issue for organizing the cleaned/new branch. I also edited the top comment with the issue description.

ausshir, 2016-07-14:
Did the pull request (while I wait for kernels to compile). Will be creating the release tag and closing the branch.

closed 2016-07-14 by ausshir