UofSSpaceTeam/roveberrypy #11: Post-URC 2016 Release Tasks

ausshir, 2016-07-12:
This is to be done on a different branch from the cleanup branch. We will be nuking all of the old code basically!
- [ ] Remove Embedded modules to its own repository. Make sure to reference the release here in the main README.
- [ ] Find out what to do with the navigation and libusst folders  @LiamBindle?
- [ ] Remove the Intel Edison image - later we will create a repo for Linux/Kernel development
- [ ] Create a barebones version of the Python 2 Rover code and move it to the legacy software repository

LiamBindle, 2016-07-12:
Is the main rover framework that StateManager? I was expecting it to be larger. Also I'm not very well versed in Python 2->3 but do you know if it doesn't work? I don't see anything in StateManager.py that obviously wouldn't work in python 3. The main.py just need's the print statements changed.

ottopasuuna, 2016-07-12:
I'm not seeing much that shouldn't work either. Main syntax issues are print statements, and the range function no longer produces a list, so you have to do stuff like this: 
```python
for x in list(range(0,MAX)):
    doThing(x)
```
The biggest thing will probably be that some libraries may need to updated or changed

ausshir, 2016-07-12:
I think that stuff should be on issue #9!

@LiamBindle had mentioned creating some new repositories to split stuff up and make merges a little easier! Here's what I propose (with his input)

Rover projects
----------------
**rover-framework** will continue to have the python code to coordinate all of the rover systems, as well as the communication modules.  Also may have some specific non-python parts that run on the main rover computer. *renamed from rover-software*

**rover-embedded** will have all of the embedded module's code, with the idea that we want to share as much code as possible between them.

**rover-os** will have any specific stuff for linux on the rover (like if we want to use bitbake or build custom device trees etc). Should also have an install script to transfer over everything relevant to the rover or generate a cramfs image to flash! Might also contain low-level code.

**rover-gui** will have any front facing GUIs *renamed from gui-software*

**rover-hardware** will contain all the PCBs in KiCad

**rover-mechanical** will mirror the PDM with stl files at major commits, as well as track mechanical issues (potentially, I have no idea)

General projects
------------------
**usst-balloon** complete HAB stuff
**usst-docs** includes *How to Contribute* and research notes in the Wiki.

Public projects to maintain
-----------------------------
**usst-electronics** should have our PCB component library, since KiCad supports loading direct from a github repo.

**usst-sbp** (or something like that) will have our communication protocol. This is @LiamBindle 's pet project! ;) Make it so we can clone this repo into a libraries directory and have it work (maybe needs an installer script).

**usst-navigation** contains our navigation/AI code if we ever make it good!

ausshir, 2016-07-12:
**We should not rename gui-software and rover-software until we are done with the cleanup issues**

ottopasuuna, 2016-07-12:
> ...Here's what I propose (with his input)

That looks like a reasonable way to split the repos up.

ausshir, 2016-07-14:
Just an idea: remove the branches of the ERC/URC2015 and URC2014 code. I don't see a need to keep it around since we have release tags.

I suggest that if we want to create patches we should do it to a new copy of the code in the legacySoftware folder in this repository.

ottopasuuna, 2016-07-16:
Have we ever actually patched old software versions? It seems to me that we just work on the current version, and if we need code from previous years, we checkout that version and copy stuff over. I could be missing something, but I'm not sure what the point of having the legacySoftware folder is.

ausshir, 2016-07-17:
Yeah I actually created a new repository for that called usst-legacy.

The repo-reorg branch will become main and dev once I have the documentation ironed out. I have to catch up! :)

ausshir, 2016-07-18:
Closing issue, see the new repositories for where things have been moved and the repo-reorg branch for where the cleaned repository code is.

closed 2016-07-18 by ausshir