UofSSpaceTeam/roveberrypy #14: Update old incompatible python 3 code in statemanager, roverprocess and main

ausshir, 2016-07-14:
Only goal is the code needs to not crash when you do a pull request! Doesn't have to do anything fun.

- [ ] main
- [ ] satemanager
- [ ] roverprocess
- [ ] Brag that we have updated to the most cutting edge standards!

May want to put this on the same branch as the server updating branch.

ottopasuuna, 2016-07-17:
I got the core rover running in python 3. Just fixed the print statements, a few changes with the semantics of including modules, and stopping the ReceiverThread in the RoverProcess.
Is this ready for a pull request, or should we wait until things like the CAN server are ported as well?

ausshir, 2016-07-17:
I did it last night too with the standard 2to3 script in tools. Does yours crash on closing?

ausshir, 2016-07-17:
Also I'd keep this branch open until the end of the milestone

ottopasuuna, 2016-07-17:
No it doesn't crash; it looks like python 3 depreciated `thread._Thread__stop()` so I changed the cleanup method to use join() instead.

ausshir, 2016-07-17:
Great! We'll hold onto this branch and see if we can port the servers or rewrite them. Would you be able to do a brief writeup on what you did for the wiki for future reference?

ottopasuuna, 2016-07-17:
Sure. Should I just make a new page called Porting to Python 3 or something like that?

ausshir, 2016-07-17:
Yeah whatever you like - you may want to create some sections. (This is more so for issue #9)

closed 2016-07-24 by ausshir