UofSSpaceTeam/roveberrypy #15: Port CAN server to Python 3 Socket module

ausshir, 2016-07-14:
See: https://docs.python.org/3/library/socket.html#socket.AF_CAN

We want to potentially use the standard module for CAN and sockets rather than use a legacy library.

This is the current library: https://python-can.readthedocs.io/en/latest/

ottopasuuna, 2016-07-24:
I found some slides from a talk for some linux conference. It's in German, but they have some code examples in C and Python.
https://glt14-programm.linuxtage.at/attachments/151_GLT2014-CANBus.pdf
May not be helpfull, but just thought I'd share.

ausshir, 2016-07-24:
Thanks, those code examples are exactly what I was looking for.

Also enjoying the beautiful strawberry shortcake at the end! :P

ausshir, 2016-09-10:
We are not focusing on the CAN server at this time. Closing for now - can be reopened as needed.

closed 2016-09-10 by ausshir