UofSSpaceTeam/roveberrypy #17: Pick a license for public code

ausshir, 2016-07-16:
Since we are going to make this code publicly available, we are going to need a license!

I am proposing the Apache 2.0 License
http://www.apache.org/licenses/LICENSE-2.0

Open to discussion!

ottopasuuna, 2016-07-16:
There is also an Educational Community License, that is basically the Apache 2.0 with a modification to the scope of patent grants.
[https://opensource.org/licenses/ECL-2.0](https://opensource.org/licenses/ECL-2.0)
I'm a fan of BSD 3-clause and GPL, but Apache seems appropriate for this project.

ausshir, 2016-07-17:
My concern with GPL is to not tie the hands of future teams and team members who may want to use a derivative of the current software but might not be able to provide sources.

ECL looks great! I always liked the simplicity of BSD. I don't think it provides much protections though. It's better than no license though!

ottopasuuna, 2016-07-17:
I think the most notable thing the Apache 2.0 has over BSD 3-clause is if you make changes to an Apache licensed work, you have to state any major changes you made. BSD does state that there is no warranty, and that we aren't liable for anything that goes wrong if you use our software. Pretty much all the protection you need, if you don't care how third parties use your software.
I agree about the GPL issue, if future USST members want/need to make a closed source derivative, they should be able to.

ausshir, 2016-07-17:
I'll bring this up on Slack with the other team leads and see what feedback I can get on the ECL.

If everyone agrees on that we can add the licenses to repo-reorg and close this issue.

closed 2016-07-18 by ausshir