UofSSpaceTeam/roveberrypy #21: Arm communication

ottopasuuna, 2016-09-07:
Communication between the arm board and main rover computer.
Be carefull what gets recycled from last year. Something was not quite right with it.

- [ ] Translate joystick input to commands and send to the arm (python side)
- [ ] Recieve commands from the rover (arm side)
- [ ] Send feedback to rover (arm side)
- [ ] Do things with the commands (turn on motors, run inverse kin, etc)

LiamBindle, 2017-01-27:
I think that we should probably get working on the arm soon -- anyone able to give me a hand? I'm not too sure how much I'll be able to work on it this sem so if someone would like to work on it with me that would be great. **Anyone**, first years too! Let me know if you're interested!

ausshir, 2017-01-27:
@LiamBindle  - Do you have any experience with Matlab SimuLink? It seems to be something that would be really valuable to simulate the arm before it is built.

That said, the mechanical team is doing a design review of the arm this Saturday so it would be good for interested people to go to it and see what it's all about.

closed 2017-07-21 by ottopasuuna