UofSSpaceTeam/roveberrypy #30: Make RecieverThread in RoverProcess not a daemon

ottopasuuna, 2016-09-20:


ausshir, 2016-09-20:
Is there a drawback for it being a daemon? I believe it's necessary for it to exit when the parent is closed with a keyboard interrupt (of course I'm sure there's a workaround).

ottopasuuna, 2016-09-20:
I was under the impression that daemon threads could continue running after the parent exited. The documentation states: 

>     The entire Python program exits when no alive non-daemon threads are left.

So my thinking was that if all the threads were daemons, the main program would exit, but the other threads could keep running without anyone knowing. By them not being daemons, you would know that they exited properly if the whole program exited.

(I was mainly trying to find a small bug that I could fix quickly for the purpose of making a git workflow demonstration).

ottopasuuna, 2016-10-26:
Turns out I mis-misread the documentation. Python docs are easy to read, hard to follow sometimes.

closed 2016-10-26 by ottopasuuna