UofSSpaceTeam/roveberrypy #41: Remove OS images from git history

ottopasuuna, 2016-11-03:
We used to keep some operating system images in this repo, but although they are not currently in the repo, they remain in the history, resulting in long download/clone times. I haven't checked, but it may be possible to remove these files throughout the entire history of the project.

ottopasuuna, 2017-05-24:
Haven't looked too closely yet, but this might do the trick: http://www.catb.org/esr/reposurgeon/

jpas, 2017-05-27:
[BFG Repo-Cleaner] is also another tool that could potentially be used

Here is a [list of all files] in the repository sorted by decreasing file size.

[BFG Repo-Cleaner]: https://rtyley.github.io/bfg-repo-cleaner/
[list of all files]: https://github.com/UofSSpaceTeam/roveberrypy/files/1033811/files.txt