UofSSpaceTeam/roveberrypy #53: Implement VESC packet handler

ottopasuuna, 2017-01-19:
We need a more robust way of encoding and decoding messages using the VESC communication protocol. @LiamBindle has developed a library that is almost complete (found here [https://github.com/LiamBindle/vesc](https://github.com/LiamBindle/vesc)). Once he is finished, it can be integrated into the USBServer.

LiamBindle, 2017-01-27:
Almost ready for PyVESC v1.0.0 release. Just need a bit more testing. Also today I cleaned up the interface a bit. Also implemented the Setter's you listed above.

ottopasuuna, 2017-02-04:
I'd say this is done. Any issues with the library its self can be done from Liam's official repo: [Pyvesc](https://github.com/LiamBindle/PyVESC/). Any new messages can be added to the USST's fork: [https://github.com/UofSSpaceDesignTeam/PyVESC](https://github.com/UofSSpaceDesignTeam/PyVESC)

closed 2017-02-04 by ottopasuuna