UofSSpaceTeam/roveberrypy #60: Class multiprocessing.boundedsemaphore not found in the python 3.6.0 library

wjtan360-git, 2017-01-30:
The class multiprocessing.BoundedSemaphore is not found when trying to import it in the ArmProcess.py file.

Multi processing bounded semaphores should be used to make sure that when one command is executed, other commands do not conflit with it.

I am going to leave this issue here so if someone comes across a solution to this problem or finds the file on the internet or from a previous version of python 3 they can notify us here. I did look in the [zip installation file of python 3.6.0 from their website](https://www.python.org/downloads/release/python-360/) and it does not contain a library that includes bounded semaphores.

With the possibility of the class not being found somewhere on the internet or in previous versions of python 3, should we make our own class for bounded semaphores?

ottopasuuna, 2017-01-30:
This is just in Visual Studio right?
On my machine it works fine:
```
Python 3.6.0 (default, Jan 16 2017, 12:12:55)
[GCC 6.3.1 20170109] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from multiprocessing import BoundedSemaphore
>>> sem = BoundedSemaphore()
>>> sem
<BoundedSemaphore(value=1, maxvalue=1)>
>>>
```
Can you find a way to open a python interpreter on your computer or Visual Studio and try importing that class? I can give you more detailed instructions on slack if you want. It could be that Visual studio messed up its python instalation, or that autocomplete isn't working reliably. I'm thinking it's just the autocomplete though, because BoundedSemaphore appears to be defined in the file Lib/multiprocessing/synchronize.py. Python is not like Java in that multiple classes can be defined in a single file of a completely different name.

wjtan360-git, 2017-01-30:
Hi Carl,

There is no problem with Visual Studio or the python installation and they are both working fine together the way the should. I was simply me not looking hard enough. I imported the synchonize.py file that you mentioned (in Visual Studio) and there it was.

I guess the people who are contributing to python should  update their documentation because that is what was causing this confusion, because the way they had it written in their documentation made me believe that there needed to be a file called BoundedSemaphore.py in the multiprocessing python library folder. They need to update it so that it can reflect the actual position of this class.

You can probably notice right now that I am going to be _really_ picky about code syntax as a heads up, because typing `from multiprocessing import BoundedSemaphore` means something completely different than typing `from multiprocessing.synchronize import BoundedSemaphore`. The former implies that there should be a filed called BoundedSemaphore.py whereas the latter implies that the class BoundedSemaphore is in a "package" or file called synchronize.py! :D 

Overall, it really frustrates me that the python community is forgetting to fix a simple edit to their documentation. To make matters worse, I was unable to find instructions on the internet on how to import the BoundedSemaphore class which lead me to create this issue in the first place.

closed 2017-01-30 by wjtan360-git