UofSSpaceTeam/roveberrypy #63: Watchdog timer

ottopasuuna, 2017-02-17:
The state manager process could keep track of which processes are supposed to be running and do some polling to find out whether anything has crashed. If a process is found to be unresponsive, the state manager will restart it. If this happens too many times (3 for example), the process has struck out and will not be restarted. Maybe on shutdown, the statemanager could log a warning of which processes, if any, failed in case too many messages are on screen.

ausshir, 2017-02-18:
After talking with @ottopasuuna, we have decided to try and put the watchdog inside the StateManager.

As it stands, I think the functionality will be to run an internal clock/alarm as a thread in StateManager, which would bite (raise an error) if each RoverProcess does not send it a message before the counter expires. Ideally, we would want the process to be restarted by StateManager if it looks frozen, and to dynamically disable the watchdog for a certain process as needed using the WebUI.

RoverProcesses would indicate that they are still alive by periodically calling pet() or something like that. That will need to be added as well.

ausshir, 2017-02-19:
I'm a little confused by an error I'm getting. Ideally, I don't want people to have to manually call pet() themselves, so it would be best to put it inside RoverProcess's run() loop. However, when I do this, I get the following error on the second call to self.pet()

```root                : INFO     Enabled modules:
root                : INFO     ['ExampleProcess']
root                : INFO     Registering process subscribers...
root                : INFO     STARTING: ['ExampleProcess']
StateManager        : INFO     StateManager shutting down
StateManager        : DEBUG    Watchdog: Timer 0 Guarding {}
StateManager        : INFO     StateManager shut down success!
Process StateManager-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/mnt/c/linux/roveberrypy/roverprocess/RoverProcess.py", line 66, in run
    self.pet() #why does it not work here?
  File "/mnt/c/linux/roveberrypy/roverprocess/RoverProcess.py", line 115, in pet
    self.publish('wd_pet', self.__class__.__name__)
  File "/mnt/c/linux/roveberrypy/roverprocess/RoverProcess.py", line 96, in publish
    self.uplink.put(RoverMessage(key, value))
AttributeError: 'dict' object has no attribute 'put'
```

I don't have this issue when calling self.pet() inside ExampleProcess's loop() function

ottopasuuna, 2017-02-19:
Unfortunately the Statemanager is a little weird in that its uplink isn't a queue like all the other process, it is instead a dictionary of queues indexed by process name. It is populated in the main.py and passed in on system startup. It might be nice to change this at some point. The easiest thing to get around this would be to check the name of the class at runtime using `self.__class__.__name__` and ignore the call if it is the StateManager.

Also, which version of python are you using? Your traceback references python 2.7 :)

ausshir, 2017-02-19:
Yes that workaround seemed to work.

Well, the code seems to still be 2-3 compatible ;)

Still ToDo: Add in mechanism to attempt to restart processes when they timeout. (Aka the hard part).

ausshir, 2017-04-07:
The watchdog timer is preliminarily completed. The following functionality is available:

- A Watchdog thread runs in the StateManager. This keeps track of running RoverProcesses and counts to a timeout if the process does not notify it (pet). The Watchdog maintains its own state dictionary in the form of {RoverProcessClassName: IsRunning}
- On each subscription registration, StateManager adds the process to the Watchdog's running state
- RoverProcesses have an automatic pet at the beginning of each loop
- RoverProcesses also have the ability to Extend and Revert the Watchdog timer, manually Pet the Watchdog, and manually Reset the entire Watchdog state (for ex. on behalf of another RoverProcess).

When a process times out, the Watchdog places the process name into a Queue of hanging processes. The parent consumes this Queue and terminates and restarts the process by name. This hanging list replaces the time.sleep() in the main loop of main.py.

By default, all processes should work with the Watchdog automatically without change (though I haven't tested it all that much).

See the attached log for what all this looks like! :)

[log.txt](https://github.com/UofSSpaceDesignTeam/roveberrypy/files/907024/log.txt)

closed 2017-04-08 by ottopasuuna