UofSSpaceTeam/roveberrypy #74: Roveberrypy doesn't work on Windows...

ottopasuuna, 2017-04-29:
Windows and Unix have different ways of handling processes which I thought should have been handled by the Python multiprocessing library, but apparently things aren't working on Windows. This isn't a total roadblock as the rover runs Linux, and most people use Cygwin, Bash on Windows, or a Unix based operating system, but it is still not ideal that rover development can't be done natively on Windows.

ausshir, 2017-06-17:
Have been looking into what is causing this. The traceback suggests that the Windows version of the multiprocessing wrapper is choking on pickling an RLock object.

The exact commit where this regression occured was 866bc5168aebdf4c67930fbe7f079c8e943fb1d4.

Initially I thought it was related to logging - and while it probably is as well, this would need to be fixed first! :)

ausshir, 2017-06-17:
Update: I take the above message back. After doing some testing, the issue has existed since porting the software to Python 3.  :'(

ausshir, 2017-06-17:
As a really hacky workaround, I discovered a new multiprocessing implementation using the same Python Threading API called Multiprocess.

https://github.com/uqfoundation/multiprocess

They also created a better pickling method called Dill which is used to pass methods and locks etc. between Windows processes in this case.

On an initial test on the first Python 3 commit, de795b2e4741a4194dc64ba4f6e7db9abaf92c86, this fixed the problems.