UofSSpaceTeam/roveberrypy #75: StateManager sometimes has no stateSem attribute

ottopasuuna, 2017-04-30:
Occasionally I get the following error on startup:
```
Exception in thread Thread-5:                                                                                                                                                                  
Traceback (most recent call last):                                   
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner                                                                       
    self.run()                         
  File "/home/carl/Programming/usst/roveberrypy/roverprocess/RoverProcess.py", line 61, in run
    self._parent.messageTrigger(message)                  
  File "/home/carl/Programming/usst/roveberrypy/roverprocess/StateManager.py", line 173, in messageTrigger
    self.addSubscriber(message.data[0], message.data[1])                                         
  File "/home/carl/Programming/usst/roveberrypy/roverprocess/StateManager.py", line 105, in addSubscriber
    with self.stateSem:                                    
AttributeError: 'StateManager' object has no attribute 'stateSem'
```

Everything seems to continue to run afterwards, but with the watchdog changes, the watchdog timer doesn't seem to be watching any processes when this happens and it prevents the StateManager from shutting down.
It appears to happen randomly, so I don't know how to reliably replicate it.

ausshir, 2017-05-01:
Is it possible that addSubscriber() is called externally before setup() due to some race condition when starting all the processes? That would make self.stateSem not exist and crash that instance of the StateManager

(If that makes any sense)

irene000, 2017-05-01:
Idea: all the subscriber is put in the uplink, which is listened by the Receiverthread of StateManager. The problem might be solved if we do the setup(which initialize stateSem) before starting the Receiverthread.

ottopasuuna, 2017-05-01:
Yeah, now that you guys mention it I'm sure that's what is causing it.

closed 2017-05-02 by ottopasuuna